linshuqin329/UPS-Lite #3: Is there a way to trigger a boot when external power is applied?

traindriverrev, 2019-08-15:
Is there a way to trigger a boot when external power is applied?

THX723, 2020-09-20:
@traindriverrev Were you able to get anywhere with this?

akula3, 2020-09-25:
I'M also interesed in this function

THX723, 2020-09-26:
Under normal operation, the answer is unfortunately NO.

However, there is a corner case in which it would work. That is if the battery is allowed to run down completely (<1% SOC), where the on-board circuit cut-off occurs. Only then will the Pi power up again, when external power is re-applied to the UPS-Lite input. That was meant as a safety mechanism to protect the lithium battery. I also wouldn't recommend relying on that method, as running lithium batteries down to 1% is extreme hard on its life. For extended service life, keep lithium operating SOC 80%-20% (70%-30% even better).

Rumor has it a newer hardware version is being worked on and would have the behavior we're looking for.

akula3, 2020-09-27:
THX723, can you please post a link to you mentioned hardware?