linshuqin329/UPS-Lite #4: Bigger battery

N2Network, 2020-01-10:
I swapped out the battery for a 10,000mAh battery. Curious if it's possible to change the capacity in the code so it reports correctly?

Fidelity88, 2020-02-08:
I think you can make adjustments in the python script. Otherwise in the manual is something about a serial connection to probably re-program the board. See #1 for the partially translated English manual.

Also, do you have a link to the battery you used?

LMoesman1998, 2020-09-14:
> I swapped out the battery for a 10,000mAh battery. Curious if it's possible to change the capacity in the code so it reports correctly?

Could you tell me what battery you used? Want to upgrade the battery as well, but not sure what to look for.