linshuqin329/UPS-Lite #5: Detecting external power

briggySmalls, 2020-02-11:
The schematic seems to indicate `Vin` is connected to `GPIO7`.

Inspecting the pogo pins it looks like something is connected to `GPIO4` or header pin number 7. So I assume this is what is meant?

However when I read this pin as an input I don't seem to get anything corresponding to Vin.

Can I detect the presence of external power with the UPSLite?

briggySmalls, 2020-02-11:
For anyone else looking into this issue:

I read the English-language translation of the manual #1 and found:

> Also UPS-Lite insertion detecting function with a power adapter, the insertion of the power pi io4 (BCM number) detects the high level, when pulled low, enabling the weld shorting function requires two back of the UPS disc, as shown below in detail.

And translating the correcsponding text in the new manual (v1.2):

> In addition, UPS-Lite has a power adapter insertion detection function, which can increase the high and low levels of the GPIO port to determine whether an external power supply is plugged in. The io4 (BCM number) of pi will detect a high level when plugged in, and low when unplugged. Level. To enable this function, you need to short the two pads on the back of the UPS. See the figure below for details.

With the diagram:

[image]

This made me realise I had not read the schematic correctly, there are clearly two pads labelled:

![Screenshot 2020-02-11 at 11 35 44](https://user-images.githubusercontent.com/11054133
/74233712-1a316580-4cc3-11ea-9c19-295af946137b.png)

Whilst I have not yet tested this, it looks like this will solve my issue.

briggySmalls, 2020-02-12:
It works!

closed 2020-02-12 by briggySmalls

lgbrownjr, 2020-06-03:
> It works!

Can you please explain how you accomplished this?  I see the two pads, but due to lack of experience, i want to be sure before potentially destroying my UPS.  Did you simply solder the two pads together?
Also, as i understand, gpio4 is also responsible for providing the battery capacity, so how does shorting the two pads affect the values?

Thanks!

briggySmalls, 2020-06-03:
## TLDR;

Yes, just solder the two pads together.

## Explanation

The effect of that is to make an electrical connection between `PAD1` and `PAD1_2` in the schematic that I posted above. Otherwise `PAD1` is just floating (not connected to anything at all).

As for your other question, it is worth being explicit about which [pin number system](https://pinout.xyz/pinout/i2c#) we are using. The battery capacity is obtained by communicating with the fuel gague on the UPSLite via the I2C protocol (a pair of pins). Specifically it uses I2C1 which is located on pins with BCM numbers 2 & 3. Thats the 3rd and 5th pins if you are looking at the physical header.

So no, BCM4 (AKA physical pin 7) is not used in communicating with the fuel gague. It needs to be configured as a simple GPIO, an input.