linshuqin329/UPS-Lite #10: New order on aliexpr and had to modify the script

ser, 2022-07-09:
Hi, I've just bought UPSlite v1.2 on aliexpress and I have a small issue:
```
# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- 32 -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                   
```
It's on 0x32 not on 0x36 anymore.

I also had to modify the script to read capacity:

```
def readCapacity(bus):
        "This function returns as a float the remaining capacity of the battery connected to the Raspi UPS Hat via the provided SMBus object"
        address = 0x32
        read = bus.read_word_data(address, 4)
        swapped = struct.unpack("<H", struct.pack(">H", read))[0]
        capacity = swapped/256-100
        return capacity
```

linshuqin329, 2022-10-09:
Hello, I can post your UPS-Lite photos for me. I want to confirm whether you bought the fake UPS-Lite. Someone copied my product, and then sold on aliexpress

maxromanovsky, 2022-10-09:
@linshuqin329  I had the same issue with the second UPS-Lite I bought. I bought it on Shopee, and it was definitely fake. Wires from the battery were thin, it had spelling mistakes. On/off switch was different... I sent it back and waiting for refund. And it had the same different I2C address.

[image]

linshuqin329, 2022-10-10:
@maxromanovsky  yes, this UPS-Lite is fake, it is not my product

maxromanovsky, 2022-10-10:
@linshuqin329 where can I buy your original product?

linshuqin329, 2022-10-10:
@maxromanovsky  here. https://www.tindie.com/products/rachel/ups-lite-for-raspberry-pi-zero/

wjhrdy, 2022-10-10:
I fell for the same product [here](https://www.amazon.com/dp/B09YTTFRF6). My GPIO 4 is never reading HIGH and I cant decode the 32 bytes on the i2c block from address `0x76`.

```
from smbus2 import SMBus
with SMBus(1) as bus:
    print(bus.read_i2c_block_data(0x76, 0, 32))

> [255, 255, 120, 160, 0, 0, 0, 0, 0, 2, 255, 255, 151, 0, 255, 255, 255, 255, 255, 255, 0, 105, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255]
```

linshuqin329, 2022-10-11:
@wjhrdy I need to see the UPS-Lite physical photo on your hand. From the pictures of Amazon, I cannot judge whether your UPS-Lite is my product

wjhrdy, 2022-10-11:
[image]

wjhrdy, 2022-10-11:
[image]

linshuqin329, 2022-10-11:
@wjhrdy I confirmed that this is a fake UPS-Lite