linshuqin329/UPS-Lite #14: Does UPS-Lite support the second generation Pi Zero 2 W

fumin, 2023-12-06:
As title, specifically does UPS-Lite support this:

https://www.raspberrypi.com/products/raspberry-pi-zero-2-w/

I think the answer is yes, since:

* the form factor of the Pi Zero 2 is the same as the first version
* the Pi Zero 2 draws less than 500mA, while the UPS-Lite can provide 1.3A.

Nonetheless, it would be great if this support can be explicitly spelled out in the docs, such as in the README or [V1.2.pdf](https://github.com/linshuqin329/UPS-Lite/blob/master/UPS-Lite_V1.2_MAX17040G/Instructions%20for%20UPS-Lite%20V1.2.pdf).

linshuqin329, 2023-12-13:
@fumin  yes ，it is support  ZERO 2W

fumin, 2023-12-13:
Got you, thanks. Would be great as an advertisement if this could be spelled out in the README. : )

closed 2023-12-13 by fumin